geo-tp/ESP32-Bit-Pirate #3: ESP32-S3DevKit built

novamostra, 2025-07-31:
First of all I would like to congratulate @geo-tp for the amazing work. I believe this project has amazing potential!
Now for anybody out there who can't wait to test it on the S3DevKit (ESP32-S3 DevKitC-1 for example) and doesn't have the time to compile the binary I already did it and is available [here](https://novamostra.com/2025/07/31/esp32-bus-pirate-for-esp32-s3devkit/) for the 16MB version. I tested Wifi, Bluetooth and LED functionality and everything seems to work great.

Today @geo-tp addded to the description the ESP32-S3 Dev Kit so I believe that from the next version he will also provide the official deployed firmware images through the releases, until then you are welcome to download my compiled version.

geo-tp, 2025-07-31:
Thanks for taking the time to compile and share a ready-to-use binary for the ESP32-S3 DevKit, truly appreciated

Just one important note:
Full support for addressable LEDs (over 50 chipsets supported via FastLED) is only available if the project is compiled on Linux. This is due to cross-compilation limitations with Windows.

I'll provide an official build for the ESP32-S3 DevKit in the next release, which should be out in a few days.

Thanks for your support.

novamostra, 2025-08-01:
> Thanks for taking the time to compile and share a ready-to-use binary for the ESP32-S3 DevKit, truly appreciated
> 
> Just one important note: Full support for addressable LEDs (over 50 chipsets supported via FastLED) is only available if the project is compiled on Linux. This is due to cross-compilation limitations with Windows (Just see that the S3 devkit build was compiled with Linux, so it should have full addressable LEDs support)
> 
> I'll provide an official build for the ESP32-S3 DevKit in the next release, which should be out in a few days.
> 
> Thanks for your support.

Yes I read your documentation and if you see the post I compiled the code in Linux. So the compiled version provided supports addressable LEDs (which I tested on my own S3-DevKit)

Of course the official build is always welcome! :)

Thank you for the great effort!