geo-tp/ESP32-Bit-Pirate #7: Perhaps choose a different name?

RoganDawes, 2025-08-05:
Hi,

While I love that you are making hardware projects and making them available for others to benefit from, I think that using the Bus Pirate name (presumably without approval) is going a little too far. Could you perhaps come up with an alternate name, perhaps "ESP32 Buccanneer", for example, that pays homage to the inspiration, while not leading to confusion?

Thank you.

closed 2025-08-06 by RoganDawes

Strykar, 2026-02-08:
@geo-tp It might not hurt to have a word with [Ian](https://forum.buspirate.com/t/have-you-seen-this-esp32-port-of-bus-pirate/1293/6?u=strykar)? 

BP accessories are very useful for tinkerers and well made, there's synergy for both projects over interoperability.

RoganDawes, 2026-02-19:
Seems like Ian agrees the name is ["a bit infringy"](https://forum.buspirate.com/t/have-you-seen-this-esp32-port-of-bus-pirate/1293/6), especially now that there are hardware plans. Suggest "ESP Pirate" as an alternative.