geo-tp/ESP32-Bit-Pirate #8: esp32c3 or esp32c6 support

skylin008, 2025-08-06:
Thanks for shared this impressive project.Does it can be support esp32c3 chip or esp32c6 chip?

geo-tp, 2025-08-06:
I don’t know the ESP32 C3 particularly well, but from what I see, it uses a completely different architecture, has less RAM, and lacks some features compared to the S3.

So no, the project won’t work as-is, it would probably require significant adaptation, it may not have the on the fly mapping capabilities that the S3 has

However, it’s very easy to adapt to almost any ESP32 S3 variant

warped-pipe, 2025-08-30:
> it’s very easy to adapt to almost any ESP32 S3 variant

i'm sorry, may i ask what all goes into making it work on other esp32s3 devices? just mods to platform.io? i would really enjoy being able to use it on the T-Deck

AndreiVladescu, 2025-08-30:
The `platformio.ini` contains configurations for each kind of board. This means build flags, pinouts for each peripheral, such as a display that is connected for one device differently than another device. Also, some boards have additional libraries from their developer, which need to be included in order to work as expected.

jakobi, 2026-02-22:
Just a note: Maybe it's time to reconsider the c-series as proper target (assuming that a single core is sufficient): the esp32-c series restored psram support with the recent esp32-c5, see eg.  waveshare "esp32-c5-wifi6-kit-n16-r8".