geo-tp/ESP32-Bit-Pirate #23: cheap yellow display

rellis-erigon, 2025-08-27:
I am a NOOB when it comes to coding.. however it would be amazing if this project could work on the cheap yellow display with an interface :-)

marcojk, 2025-08-27:
If you are referring to the ESP32-based ones: no way. PoorMCU performance and little to no GPIO pins available.
For the bigger screens based on ESP32S3 there is hope if the display controller is supported by TFT_eSPI library...

geo-tp, 2025-08-27:
> If you are referring to the ESP32-based ones: no way. PoorMCU performance and little to no GPIO pins available.
> For the bigger screens based on ESP32S3 there is hope if the display controller is supported by TFT_eSPI library...

This