geo-tp/ESP32-Bit-Pirate #27: Web Flashing Doesnt Work

mbt28, 2025-09-05:
Hello

Web flashing for esp32-s3 devkit doesnt work. I checked the webflasher source code and it seems binary file is missing in firmware folder.

https://geo-tp.github.io/ESP32-Bus-Pirate/webflasher/firmwares/bus_pirate_08_s3devkit.bin

mbt28, 2025-09-05:
Hello,

Thank you for quick fix however I still cannot upload it. It says prepearing the installation however nothing happens. I use chrome on raspian. 

```
uname -r
6.12.42-v8-16k+

lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

I was able to flash it with success using esptool however the application doesnt work:
`sudo /home/$USER/esptool/bin/python3.11 /home/$USER/esptool/bin/esptool.py  --chip esp32s3 --port /dev/ttyACM0 --baud 460800 write_flash -z 0x0 bus_pirate_08_s3devkit.bin`

I am getting:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
```

geo-tp, 2025-09-07:
it seems the same issue as here : https://github.com/geo-tp/ESP32-Bus-Pirate/issues/11

You probably don't use the correst USB port

mbt28, 2025-09-08:
Hello,

Indeed it works with native usb port! Thank you so much for your help, I cant wait to try it. :)