geo-tp/ESP32-Bit-Pirate #40: XIAO ESP32-S3

carlmbecker, 2025-09-16:
Hello, 
Programed the device (XIA ESP32-S3) with web flasher and then went to the terminal. Get the menu but when try to put into one of the modes always responds with "Type 'help' or 'mode' but never enters any of the modes.
thanks.

geo-tp, 2025-09-16:
[image]

I'm not sure I understand your problem, you have to type `mode` and select the index number or  `mode I2c` for example to change mode (the capture is from Xiao ESP32S3)

carlmbecker, 2025-09-16:
Hello Geo

I will load into the IDE as a project when I get a chance and flash from the IDE but wanted to test this out quickly so programmed the ESP32 from the web programmer. The app does not work but continually tells you to enter 'help' or 'mode' every time a key us pressed see below : 

[image]

geo-tp, 2025-09-16:
You seem to have a problem where your terminal app does not support carriage returns, `mode 1` will not work no matter what, it is either you use `mode` alone and then you can prompt the index, or `mode spi` directly

carlmbecker, 2025-09-16:
I quickly reprogrammed the device and as you can see above (Murphy's Law) behaved differently.

carlmbecker, 2025-09-16:
Tried another terminal app called QComm and same issue. It must be a issue with either web programmer or code. Will look when I get a chance.

[image]