geo-tp/ESP32-Bit-Pirate #43: Issues at SUBGHZ load command

pecaxxl, 2025-09-21:
Hello,

First of all, I would like to sincerely thank you for implementing the .sub file send function. I hope that other users will find it useful as well. I had attempted to develop this feature myself, but due to limited time and experience, I was unable to make significant progress. I only managed to implement file upload and RAW protocol transmission, and even that was not nearly as clean or well structured as your implementation, particularly when it came to the controller–service–formatter approach.

I found few issues with load command file parsing and sending. I will try to explain best I can. 

1. When sending a file with Protocol: RAW for example, [AVANTEK_Mini_CB11_Single.sub](https://github.com/Zero-Sploit/FlipperZero-Subghz-DB/blob/main/subghz/Doorbells/AVANTEK/AVANTEK_Mini_CB11_Single.sub)

Wave form looks like this:
[image]

Zoomed in, the waveform appears uniform (no gaps, no symbols):
[image]

Below are screenshots from the FlipperLAB Pulse plotter for the same file and the URH waveform produced by my sendfile function(almost same timing):

[image]

[image]

2. When sending a file with Protocol: BinRAW for example, [DirecTV_RF_2.sub](https://github.com/Zero-Sploit/FlipperZero-Subghz-DB/blob/main/subghz/Cable_Satellite_Boxes/DirecTV/DirecTV_RF_2.sub)

File contains only one frame but is parsed as 3 frames:

[image]
Frames 2. and 3. contains no data and can't be sent.
Waveform of frame 1. looks like this:

[image]

No meaningful data is transmitted. I checked with other files containing more data, but the transmission is always incorrect.

File [DirecTV_RF_0.sub](https://github.com/Zero-Sploit/FlipperZero-Subghz-DB/blob/main/subghz/Cable_Satellite_Boxes/DirecTV/DirecTV_RF_0.sub) which contains 2735 bits and there is a wave form:

[image]

3. When sending a file with Protocol: Princeton for example, [Doorbell0.sub](https://github.com/Zero-Sploit/FlipperZero-Subghz-DB/blob/main/subghz/Doorbells/Doorbell0.sub)

File contains only one frame but is parsed as 2 frames:

First frame is invalid, all zeros:

[image]

Second frame is good and decoded as "Key:" field in file:

[image]

Protocol Holtek files with a single frame are also parsed as two frames.

I tried to find the problems but didn’t succeed. I hope this will help you identify them. Thank you very much for the time, effort, and dedication you’ve put into creating and improving this great tool.

P.S. Is the CAME protocol not implemented yet?

geo-tp, 2025-09-21:
Thanks for the detailed report again, I'll take a closer look soon to try fix that

geo-tp, 2025-09-22:
I’ve pushed some fixes that should solve most of these issues. I don’t have the right tools to analyze the signals and see exactly what’s happening, so if you could confirm by testing, that would be great

pecaxxl, 2025-09-22:
Thank you. I can confirm that RAW protocol data transmission is working correctly. I’m still unsure whether every Data_RAW line in a BinRAW protocol file should be treated as a separate frame, or as a single burst like in a RAW protocol file. There is a [Flipper zero Docs](https://developer.flipper.net/flipperzero/doxygen/subghz_file_format.html) for reference.

I don’t have the proper equipment to fully test all supported protocols, but I’ll check as much as I can with what I have available. If someone with a Flipper Zero could compare the waveforms or share a waveform of a known signal from the database that would be very helpful to me for cross-checking results.

I’ll continue testing other protocols tomorrow evening and will share any findings here.

closed 2025-09-29 by geo-tp