geo-tp/ESP32-Bit-Pirate #46: Cannot setup bus pirate on a esp32-S3

askondro, 2025-09-24:
I tried the webflasher and it reports as successful. But it stuck at th bootloader.

HW data:

start detect chip...please wait
chip sync ...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: Wi-Fi, BT 5 (LE), Dual Core + LP Core, 240MHz, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: ecda3b678a4c
Manufacturer: 5e
Device: 4018
Status value: 0x100200
Detected flash size: 16MB 

When connection with a serial application (putty) it only shows:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

What am I doing wrong?

geo-tp, 2025-09-24:
https://github.com/geo-tp/ESP32-Bus-Pirate/issues/11

Should be related to that, try with the other USB port

AndreiVladescu, 2025-09-24:
Hello! Your ESP32-S3 boots into the app successfully, the problem is that the ESP32 Bus Pirate waits for the JTAG/USB interface of the ESP32-S3 to be connected to the PC, not a USB-to-TTL converter, as you have in the picture.

What you can do to make this work is: wire a USB A male connector with IO19 to D-, IO20 to D+, 5V to 5V and GND to GND, this is the pinout for the USB connector: 

[image]

fdufnews, 2025-09-27:
'Also, if you want to use Wifi, you need to use an ipex antenna with this board,'

I have one of this boards and it also has a PCB antenna (wich is selected by default, in my case).
There is a jumper (red arrow in the image) that you can move to select the PCB antenna or the IPEX one. It is a bit tricky to move the jumper as it is a very small CMS component.
The IPEX antenna usually performs better than the PCB one.

[image]

closed 2025-09-27 by geo-tp

askondro, 2025-10-06:
Version 1.1           Ready to board

 Type 'mode' to start or 'help' for commands

HIZ> m

Select mode:
  1. HIZ
  2. 1WIRE
  3. UART
  4. HDUART
  5. I2C
  6. SPI
  7. 2WIRE
  8. 3WIRE
  9. DIO
  10. LED
  11. INFRARED
  12. USB
  13. BLUETOOTH
  14. WIFI
  15. JTAG
  16. I2S
  17. CAN
  18. ETHERNET
  19. SUBGHZ
  20. RFID
  21. RF24

Mode Number > 10

Mode changed to LED

LED Configuration:
Data pin [35]: 33
Clock pin [36]:
[WARNING] Data pin cannot be changed. Data pin set to : 35
Number of LEDs [1]:
Brightness (0–255) [128]:
Current protocol: 'ws2812'
You can change it with 'setprotocol'
or try to detect it automatically with 'scan'
LEDs configured.

Why is not possible to change the LED pin?

fdufnews, 2025-10-06:
> Why is not possible to change the LED pin?

The Pin number for the LED is set at compile time and cannot be changed later. It is imposed by the library used to manage the LED.

geo-tp, 2025-10-07:
If one of you can make an env (platformio.ini) suitable for this type of board, I will add the bin to the webflasher

fdufnews, 2025-10-07:
Yes you could, the changes I made are:
- adding GPIO 35,36,37 to the PROTECTED_PINS list
- and setting the GPIO to my liking

The devkit with and without PSRAM are the same except for the 3 forbidden GPIOs.

fdufnews, 2025-11-20:
> Has anyone used this generic ESP32-S3 for this project before?

Yes, I have two of theese an ESP32-S3-N16R8 and an  ESP32-S3-N8R2.
You must use the USB C - OTG socket to connect the board and choose ESP32-S3 DevKit on the WebUI.